drakkar-lig/walt-python-packages #2: Exception when cloning an image already cloned by another user

jonglezb, 2017-09-19:
One image is available either locally (another user already cloned it) or from the docker hub:
```
$ walt image search minimal
User           Image name                              Location     Clonable link                                               
-------------  --------------------------------------  -----------  ----------------------------------------------------------  
davtyannarek   pc-x86-64-minimal-netconsole            WalT server  server:davtyannarek/pc-x86-64-minimal-netconsole            
jonglezb       pc-x86-64-minimal-netconsole            docker hub   hub:jonglezb/pc-x86-64-minimal-netconsole                   
```

When I try to clone it from the hub:
```
$ walt image clone hub:jonglezb/pc-x86-64-minimal-netconsole
An image has the same name in your working set.
This operation would overwrite it.
If this is what you want, rerun with --force
$ walt image clone --force hub:jonglezb/pc-x86-64-minimal-netconsole
hub api_call: CSAPI clone_image ('hub:jonglezb/pc-x86-64-minimal-netconsole', True) {}
clone workflow is: save_initial_images, verify_overwrite, remove_ws_image, pull_hub_image, workflow_if_instance, update_walt_image
Downloading jonglezb/pc-x86-64-minimal-netconsole... done.
server-blocking: Exception occured while performing API request:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/walt/common/thread.py", line 174, in handle_api_call
    res = local_service.do(path, args, kwargs)
  File "/usr/local/lib/python2.7/dist-packages/walt/common/apilink.py", line 75, in do
    return obj(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/walt/server/threads/blocking/thread.py", line 12, in clone_image
    res = clone(context.requester.sync, self.server, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/walt/server/threads/blocking/images/clone.py", line 287, in clone
    **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/walt/server/threads/blocking/images/clone.py", line 274, in perform_clone
    res = workflow_run(workflow, **context)
  File "/usr/local/lib/python2.7/dist-packages/walt/server/threads/blocking/images/clone.py", line 182, in workflow_run
    res = f(**context)
  File "/usr/local/lib/python2.7/dist-packages/walt/server/threads/blocking/images/clone.py", line 174, in workflow_if_instance
    if condition_func(**context) != False:
  File "/usr/local/lib/python2.7/dist-packages/walt/server/threads/blocking/images/clone.py", line 105, in verify_compatibility_issue
    ws_image = image_store[ws_image_fullname]
  File "/usr/local/lib/python2.7/dist-packages/walt/server/threads/main/images/store.py", line 77, in __getitem__
    return self.images[image_fullname]
KeyError: u'jonglezb/walt-node:pc-x86-64-minimal-netconsole'
```

jonglezb, 2017-09-19:
Same issue when cloning from the local server:
```
$ walt image clone server:davtyannarek/pc-x86-64-minimal-netconsole
An image has the same name in your working set.
This operation would overwrite it.
If this is what you want, rerun with --force
$ walt image clone --force server:davtyannarek/pc-x86-64-minimal-netconsole
hub api_call: CSAPI clone_image ('server:davtyannarek/pc-x86-64-minimal-netconsole', True) {}
clone workflow is: save_initial_images, verify_overwrite, verify_compatibility_issue, remove_ws_image, tag_server_image_to_requester, update_walt_image
server-blocking: Exception occured while performing API request:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/walt/common/thread.py", line 174, in handle_api_call
    res = local_service.do(path, args, kwargs)
  File "/usr/local/lib/python2.7/dist-packages/walt/common/apilink.py", line 75, in do
    return obj(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/walt/server/threads/blocking/thread.py", line 12, in clone_image
    res = clone(context.requester.sync, self.server, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/walt/server/threads/blocking/images/clone.py", line 287, in clone
    **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/walt/server/threads/blocking/images/clone.py", line 274, in perform_clone
    res = workflow_run(workflow, **context)
  File "/usr/local/lib/python2.7/dist-packages/walt/server/threads/blocking/images/clone.py", line 182, in workflow_run
    res = f(**context)
  File "/usr/local/lib/python2.7/dist-packages/walt/server/threads/blocking/images/clone.py", line 105, in verify_compatibility_issue
    ws_image = image_store[ws_image_fullname]
  File "/usr/local/lib/python2.7/dist-packages/walt/server/threads/main/images/store.py", line 77, in __getitem__
    return self.images[image_fullname]
KeyError: u'jonglezb/walt-node:pc-x86-64-minimal-netconsole'
```

eduble, 2017-09-19:
There are 2 possibilities:
1. it is a bug, or
2. you did something manually (not using walt commands) which made the walt database inconsistent regarding the local set of docker images: 
  2a. maybe you registered manually the image in walt database, or
  2b. you downloaded an image using a docker command (not walt image clone)

Any idea?

If it's case 2, I guess I should make it more robust by checking the local images and sync'ing the database before starting the clone.
If it's case 1, I will have to investigate more.

jonglezb, 2017-09-19:
Hmm, you're right, I had downloaded docker images outside of walt, because there was an issue with the search.

The suggestion to make it more robust has been implemented in cd3161c1fa07f753c24bf6a3207e2445ad586319 and a7436c8963486b906b26b975ebbf8ac0e951897c

closed 2017-09-19 by jonglezb