drakkar-lig/walt-python-packages #3: "walt node cp" should behave like the regular "cp" command

jonglezb, 2017-09-25:
Currently `walt node cp` works very differently from the usual `cp` command:

- it preserves user/group, which is annoying because nodes mostly run everything as root.  `cp` does not preserve user/group by default.
- it does not follow symbolic links, instead it copies them as symlink on the node
- it does not allow "-" as the name of a file (to take input as stdin or redirect output to stdout)
- it does not allow to name the file remotely, e.g. `walt node cp foo.txt mynode:/tmp/bar.txt` gives an error

Also there are some issues:

- it fails when files are large, because of a temporary copy in `/tmp/` which can fill up the filesystem there
- when it fails (e.g. with large files), it leaves temporary files in `/tmp/` and does not give any failure indication to the client.

eduble, 2017-09-26:
Here is a commit that may have introduced some of the problems you noticed: https://github.com/drakkar-lig/walt-python-packages/commit/ebc33391770614ed10f1cfdf1ac7719a8570261a

I made those changes to make `walt node cp` work when using minimal images (i.e. when using the limited busybox version of the `tar` command). The commit message explains this in more details.

I am not sure the issue you met was related to "filesystem filling". Unless I miss something, we just create a temporary symlink on the sender side, no plain copy. (Without this symlink, we might overwrite a file on the target, if it has the same name as the source.) Maybe it was another reason?
We will have to recheck all that.

Hugof1, 2017-10-09:
Hi all,
I did a preliminary diagnosis.

> it preserves user/group, which is annoying because nodes mostly run everything as root. cp does not preserve user/group by default.

I could reproduce what you describes:
```
root@rpi3-hugo:~# ls -l /tmp/myconf.conf
-rw-r--r-- 1 1003 1003 19 Oct  9 09:39 /tmp/myconf.conf
root@rpi3-hugo:~# 
```
The file I copied has user-ID `1003`, and of course this user ID does not exist on the node.

> it does not follow symbolic links, instead it copies them as symlink on the node

I could also reproduce this.
`cp` dereference links by default, `walt node cp` does not.

> it does not allow "-" as the name of a file (to take input as stdin or redirect output to stdout)

Actually, `cp` or `scp` commands do **not** handle this either! `tar` does... It could be handy in some cases.

> it does not allow to name the file remotely, e.g. walt node cp foo.txt mynode:/tmp/bar.txt gives an error

I could also reproduce this.
```
hugo@oeno:~# walt node cp myconf.conf rpi3-hugo:/tmp/testfile.txt
No such file or directory: /tmp/testfile.txt
hugo@oeno:~# 
```

The same behavior is observed when using a minimal image (busybox-based) on the node.

eduble, 2017-10-09:
> - it fails when files are large, because of a temporary copy in `/tmp/` which can fill up the filesystem there

I can confirm that there is no temporary copy. We just use a temporary symlink to manage the renamed destination file (which apparently does not work well), and Baptiste thought is was a plain file copy.

There may still be filesystem full issues when copying large file since new files are stored in the node's RAM, but this is the expected behavior of WalT nodes: in order to ensure reproducibility, we want a node to boot exactly the same OS at each reboot on a given image. If you have to store big files, you should modify the image and deploy it again.

> - when it fails (e.g. with large files), it leaves temporary files in `/tmp/` and does not give any failure indication to the client.

Right, that is something we should improve.

Hugof1, 2017-10-10:
I tested the other way (node to client), everything works!

Hugof1, 2017-10-10:
> it does not allow to name the file remotely, e.g. walt node cp foo.txt mynode:/tmp/bar.txt gives an error

This is fixed by https://github.com/drakkar-lig/walt-python-packages/commit/db9fef8aa60f8f11190c8ae7642f901ecdd551f3.

eduble, 2017-10-10:
> it preserves user/group, which is annoying because nodes mostly run everything as root. cp does not preserve user/group by default.

This is also fixed: https://github.com/drakkar-lig/walt-python-packages/commit/eb2abf440b0b7c9458647f5727796df2ce376943.

Hugof1, 2017-10-11:
I tested `walt image cp`, I saw the same behavior with symbolic links, the rest works.

eduble, 2017-10-11:
> it does not follow symbolic links, instead it copies them as symlink on the node

This is fixed by https://github.com/drakkar-lig/walt-python-packages/commit/4e7197e58e195fb7850b30a98f01c378f72e31c5.

eduble, 2017-10-16:
> when it fails [...] it [...] does not give any failure indication to the client.

This is fixed by: https://github.com/drakkar-lig/walt-python-packages/commit/5bbcbde3e117f070bb91b56a0c5a8a110f97ac3f

eduble, 2017-10-16:
> when it fails [...], it leaves temporary files in /tmp/ [...].

This is fixed by: https://github.com/drakkar-lig/walt-python-packages/commit/2705cbd04edb827e5b5aee19c320b96e220b6d1b

eduble, 2017-10-16:
All issues are solved, closing the issue.
We will keep in mind a possible improvement of being able to use `-` for stdin or stdout, as `tar` does.

closed 2017-10-16 by eduble