drakkar-lig/walt-python-packages #4: Saving an image after "walt image shell" fails with KeyError: 'status'

jonglezb, 2017-10-24:
On milo:

```
$ walt image shell pc-x86-64-debian-stretch
# Do some stuff
root@image-shell:/# exit
New image name [pc-x86-64-debian-stretch]: pc-x86-64-debian-stretch-pythonnet
Unexpected server-side issue! 'status'
```
The image is not saved.

Server log:
```
hub api_call: CSAPI create_image_shell_session ('pc-x86-64-debian-stretch', 'shell session') {}
hub api_call: CSAPI image_shell_session_save (0, 'pc-x86-64-debian-stretch-pythonnet') {'name_confirmed': False}
Exception occured while performing API request:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/dist-packages/walt/server/threads/main/task.py", line 28, in run
    res = m(self.context, *self.args, **self.kwargs)
  File "/usr/local/lib/python2.7/dist-packages/walt/server/threads/main/api/cs.py", line 193, in image_shell_session_save
    return session.save(context.requester.sync, new_name, name_confirmed)
  File "/usr/local/lib/python2.7/dist-packages/walt/server/threads/main/images/shell.py", line 58, in save
    if event['status'] == 'die' and \
KeyError: 'status'
```

This is probably caused by the newer version of Docker.

jonglezb, 2017-10-24:
I missed the docker error in the logs:
```
dockerd[3772]: time="2017-10-24T09:44:27.044286382+02:00" level=error msg="Handler for POST /v1.29/containers/d2f11ce1-f548-4782-ae3b-29fcd9abc72b/kill returned error:  Cannot kill container d2f11ce1-f548-4782-ae3b-29fcd9abc72b: Container 83a7e3102215c47e212d83bbede62a661226bdcf8527602acf06162cfc417740 is not running"
```

eduble, 2017-10-24:
On 24/10/2017 09:51, Baptiste Jonglez wrote:
> I missed the docker error in the logs:
>
> |dockerd[3772]: time="2017-10-24T09:44:27.044286382+02:00" level=error
> msg="Handler for POST
> /v1.29/containers/d2f11ce1-f548-4782-ae3b-29fcd9abc72b/kill [...]

This is not really an issue, we just call the same function to "kill", 
"stop" and "wait" for the container in various cases, and ignore errors. 
In this case, the "kill" step is useless and generates this error log, 
but it has no incidence on walt's behavior.

-- 
Etienne Dublé
CNRS / LIG - Bâtiment IMAG
700 avenue Centrale - 38401 St Martin d'Hères
Bureau 426 - Tel 0457421431

eduble, 2017-10-24:
As you guessed, the new version of Docker sends new kinds of events that we were not handling properly.
This is fixed in https://github.com/drakkar-lig/walt-python-packages/commit/389ecd51dd2803043e030c12916ef9f3553d60c5.

closed 2017-10-24 by eduble