drakkar-lig/walt-python-packages #16: Python 3 support

audeoudh, 2018-09-26:
Python2 End Of Life is 2020 (see e.g. [PEP&nbsp;373](https://www.python.org/dev/peps/pep-0373/#update)).  This is not so soon, but we have to think at porting WalT to Python3.

audeoudh, 2018-09-27:
Supporting both python 2 & 3 is theoretically possible, but may make the code more complex.  As python 3 is widely deployed on machines, we will drop the support of Python 2.

This modification will _a priori_ require the client and the server upgrades to python3 simultaneously: the object exchange through pickle seems to be difficult between versions. However, the interaction between the server and the nodes should not raise errors, so we won't require all nodes be upgraded.

eduble, 2020-09-08:
Closing: the long awaited [version 5](https://listes.univ-grenoble-alpes.fr/sympa/arc/walt-users/2020-09/msg00000.html) now runs with python3.

closed 2020-09-08 by eduble