drakkar-lig/walt-python-packages #29: full history does not match any log records when used together with a named stream

sandikkaya, 2018-11-29:
A common case of receiving collected logs in walt is checking a range of logs via `walt log` command. A typical example is: `walt log show --history -15m:` which visualizes all collected logs of the respective user in the last 15 minutes. Similarly, `walt log show --history :` or `walt log show --history full` matches the whole history of logs.

A better way to refine the presented logs is used named log streams, which could be added to the logs while they are collected. Then, one can receive the last 15 minutes of relevant logs by running `walt log show --history -15m: --streams my_log_stream`.

It is expected to receive the whole history of a log stream with a full range such as `walt log show --history : --streams my_log_stream` or `walt log show --history full --streams my_log_stream`.

However, the full history option matches none of the log records when queried together with a stream name.

sandikkaya, 2019-01-09:
I have tried both of the options that previously did not return anything. Currently, it seems like fixed. Or it was once problematic in a way we could not figured out.

eduble, 2019-01-09:
OK, so I will close it, and feel free to re-open if you see an issue again.

closed 2019-01-09 by eduble