drakkar-lig/walt-python-packages #43: Problem with walt-rpc leading to reboot

Warxo, 2019-10-25:
Hello,
I'm trying to setup walt on APU4C4 nodes.
I want to use Debian Buster, and i'm using this Dockerfile :
```
# step B.4 debian include init
FROM debian:buster

# specify which node models this image can handle
# step B.1
LABEL walt.node.models=pc-x86-64

# Install packages
# step B.5 for busybox
# step B.6 for openssh

RUN sed -i "s#main#main contrib non-free#g" /etc/apt/sources.list

RUN apt-get update && \
    apt-get install -y linux-image-amd64 systemd-sysv openssh-server \
    busybox-static locales netcat-openbsd lldpd vim hostapd firmware-atheros \
 && rm -rf /var/lib/apt/lists/* \
 && rm -rf /var/cache/apt/*

 # Unreleased klibc that fixes a bug where ipconfig does not work properly
 # with multiple interfaces (Debian bug #852480)
 ADD klibc-utils_2.0.4-10_amd64.deb libklibc_2.0.4-10_amd64.deb /tmp/
 RUN dpkg -i /tmp/klibc-utils_2.0.4-10_amd64.deb /tmp/libklibc_2.0.4-10_amd64.deb && \
     rm -f /tmp/*klibc*.deb && update-initramfs -u

# Install custom ipxe boot scripts
# step B.2 for ipxe
COPY boot/ /boot

# Install image spec file
COPY image.spec /etc/walt/

# Configure locale
RUN echo 'en_US.UTF-8 UTF-8' >> /etc/locale.gen && \
    echo 'fr_FR.UTF-8 UTF-8' >> /etc/locale.gen && \
    locale-gen

# Allow passwordless root login on the serial console
RUN sed -i -e 's#^root:[^:]*:#root::#' /etc/shadow

COPY hostapd.conf /etc/hostapd/hostapd.conf

RUN apt-get update && \
apt-get install -y iw

```

Right now the nodes are stuck in a boot loop. This is related to walt-rpc not finding the cat command, according to the boot log :

```
[   11.003395] igb 0000:01:00.0 enp1s0: igb: enp1s0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX/TX
[   11.119140] IPv6: ADDRCONF(NETDEV_CHANGE): enp1s0: link becomes ready
[   11.129416] console [netcon0] enabled
[   11.133100] netconsole: network logging started
done.
Begin: Running /scripts/init-premount ... done.
Begin: Mounting root file system ... Begin: Running /scripts/nfs-top ... done.
Begin: Running /scripts/nfs-premount ... done.
modprobe: module af_packet not found in modules.dep
IP-Config: enp1s0 hardware address 00:0d:b9:51:a5:5c mtu 1500 DHCP
IP-Config: enp1s0 complete (dhcp from 192.168.183.1):
 address: 192.168.183.2    broadcast: 192.168.183.255  netmask: 255.255.255.0   
 gateway: 0.0.0.0          dns0     : 0.0.0.0          dns1   : 0.0.0.0         
 host   : ap                                                              
 rootserver: 192.168.183.1 rootpath: 
 filename  : 
[   11.500403] FS-Cache: Loaded
[   11.577941] RPC: Registered named UNIX socket transport module.
[   11.583962] RPC: Registered udp transport module.
[   11.588688] RPC: Registered tcp transport module.
[   11.593407] RPC: Registered tcp NFSv4.1 backchannel transport module.
[   11.676402] FS-Cache: Netfs 'nfs' registered for caching
done.
Begin: Running /scripts/nfs-bottom ... done.
Begin: Running /scripts/init-bottom ... done.
*** Mounting...                    ***
*** Preparing rootfs...            ***
*** Mounting filesystem union...   ***
*** Mounting persistent filesystem ***
*** Re-mounting over the union...  ***
*** Setting clock...               ***
/bin/walt-rpc: line 6: cat: not found
[   26.301116] random: crng init done

date: invalid date '+%Y-%m-%d %H:%M:%S'
date: invalid date ''
>>> Issue detected!               <<<
Will reboot... (or press <ENTER> now for a shell)
timeout: invalid option -- 't'
BusyBox v1.30.1 (Debian 1:1.30.1-4) multi-call binary.

Usage: timeout [-s SIG] SECS PROG ARGS

Runs PROG. Sends SIG to it if it is not gone in SECS seconds.
Default SIG: TERM.
[   36.537237] reboot: Restarting system
[   36.540948] reboot: machine restart
```

Any hint on this matter ? Did I miss something in the Dockerfile ?

Thanks in advance,
Paul

eduble, 2019-10-28:
I think this means the shell of this busybox version does not recognize `cat` as an applet.

To fix this, you can try to modify `walt-init` script for this image.
Your image is mounted at `/var/lib/walt/images/<user>/<img-name>_<img_tag>/fs`, and `walt-init` should be at `/var/lib/walt/images/<user>/<img-name>_<img_tag>/fs/bin/walt-init`.

At the top, you should see this definition:
```
ROOTFS_BUSYBOX_APPLETS="
    sh ls timeout reboot sleep nc chroot
    mktemp mkfifo sed
"
```

Add `cat` to this list.

Then, you can check if the problem is solved.

For a more general and permanent fix, you have to modify the original copy of this script, which should be at:
`/usr/local/lib/python2.7/dist-packages/walt/server/threads/main/images/walt-init` (modify it the same way)
Next time an image is exposed to nodes (i.e. exported as an NFS share), this fixed version of the script will be copied there.

**Background information:**

When the walt node starts, a script called `walt-init` is run, instead of the regular init system (systemd in this case). This is the script that prints the  `*** [MESSAGE] ***` logs you see in the log dump. When all went well, at the end the regular init system (systemd) is started.

Since `walt-init` has to run on various kinds of operating systems, it has very little knowledge about the OS it runs on (except the things we wrote in the walt image requirements docs). So `walt-init` starts by creating a minimal `rootfs` based on `busybox`, and runs several subscripts in this more controlled environment. The script for setting clock is one of them.

eduble, 2019-10-28:
I forgot to mention that those modifications should be done on walt server side (because `walt-init` is a generic script, it is run whatever the image is, thus it is managed by the server).

eduble, 2019-10-29:
Please tell me if the proposed fix works. If yes, I will commit the change.
Just for my information I would also be interested if you could describe your walt platform in more details  (if you use other kinds of nodes, etc.), and if you faced some issues with the rest of the installation.

Warxo, 2019-10-30:
So, yes thanks the fix mainly worked. I had to change walt-nfs-watchdogs as well (removing the "-t" from the timeout command), probably because of a change in timeout syntax in buster (might have to change that in other scripts as well).

For now, we only test the platform, using apu4c4 nodes, so nothing particularly interesting for the moment.
I didn't face others issues in particular with the rest of the installation,

Thanks
Paul

eduble, 2019-10-30:
> So, yes thanks the fix mainly worked.

OK.

> I had to change walt-nfs-watchdogs as well (removing the "-t" from the timeout command), probably because of a change in timeout syntax in buster (might have to change that in other scripts as well).

You are true, when I type `busybox timeout --help` on buster I can confirm the syntax changed compared to stretch. The script will have to detect the correct syntax.

> For now, we only test the platform, using apu4c4 nodes, so nothing particularly interesting for the moment.
> I didn't face others issues in particular with the rest of the installation,

OK, good to hear.

FYI, we did some important changes recently, but new installation images including those changes are not published yet. I will update the platform we have here at LIG, and, if all goes well, a new version should be published by the end of the year. It should include:

* python3 support (and dropped python2 support, unfortunately we have too little manpower to maintain compatibility with both versions)
* distant nodes support: you can deploy a raspberry pi 3b+ node at home, it will setup a ssh-based vpn to reach walt experimentation network, and boot a kvm-based virtual walt node. This virtual node appears the same as virtual nodes running on the server (except the CPU architecture, ARM in this case). And since USB ports of the raspberry pi are attached to the virtual node, you can connect for instance a LoRa mote or LoRa gateway on the raspberry pi and perform interesting experiments with the same ease you get with local walt nodes.
* (not sure yet) support for aliases: a walt user could record an alias for a given device or group of device, and use this alias in subsequent commands. (note: aliases would be user-specific whereas device names are global to the platform.)

On the development side, we also automated our tests, mainly by using virtual nodes.

Warxo, 2019-10-30:
Ok, thanks for the information, good to hear.

eduble, 2020-09-08:
Closing this, since the long awaited [version 5](https://listes.univ-grenoble-alpes.fr/sympa/arc/walt-users/2020-09/msg00000.html) that we just released fixes this issue.

closed 2020-09-08 by eduble