drakkar-lig/walt-python-packages #44: Deployment question

rgrunbla, 2019-11-07:
Hello,

First, thanks for the software, we're not using it fully right now but it's already saving us time.

We want to reuse a building network to be able to easily move the nodes an vary wireless conditions. We wonder two things:

- do Walt nodes support "tagged" mode for VLANs ? The goal is having hybrids ports on the switches to be able to seemlessly change from walt nodes to "normal machines". It seems to me it's the case (as we have a vlan configuration option in the server.conf), but I prefer to be sure

- Is it possible to de-couple the DHCP functionality of the server to let some third-party manage it? We may have access to a shared VLAN where there is already some dhcp server we have to keep in charge of the dhcp part (or at least, do not disturb the machines already in there);

- Or, last, do you think it could be possible to use walt over a classic network without any dhcp at all? Could the vpn thing in preparation be used for that, maybe?

Thanks,
Rémy

eduble, 2019-11-08:
Hello Rémy

> First, thanks for the software, we're not using it fully right now but it's already saving us time.

You are very welcome, and thanks for your feedback.

Let's start with an example of what we can do. Here at LIG, our main walt platform is deployed in the IMAG building. The walt server is installed in the datacenter on the ground floor and it is connected to 3 different VLANs:
* The platform network VLAN, with id=2399 here (called `walt-net` in the documentation)
* The regular VLAN for servers (mandatory to reach internet, and more specifically the docker hub in the case of walt)
* The VLAN id=650 which allows interacting with the switches in the building (called `walt-adm` in the documentation)

Note : The server actually uses 2 physical network interfaces. One is used to connect the regular VLAN for servers (no VLAN tag). The other one connects the two walt-specific VLANs (tagged 2399 and 650) on a single cable.

Having this in place:
* We have a set of dedicated wall plugs attached to `walt-net`. You can turn a raspberry pi or a laptop into a walt node by connecting it to one of these plugs and using appropriate walt bootloader software.
* If I need one more plug in my office, I just need to ask my network admin to change the VLAN id of this plug to 2399.
* Of course, I can also disconnect one of them from its plug and replace it with a switch where I can connect more nodes.
* I can scan the network and display a network topology tree (`walt device tree`). In our case, the walt server has to send SNMP requests to various switches, many of them in `walt-adm` (switches of the building), and others in `walt-net` (because of previous point).
* I can "hard-reboot" any node because all our switches support PoE (again, the walt server uses SNMP requests to make the appropriate switch disable PoE temporarily on the appropriate port).

Back to your questions:

> * do Walt nodes support "tagged" mode for VLANs ? The goal is having hybrids ports on the switches to be able to seemlessly change from walt nodes to "normal machines". It seems to me it's the case (as we have a vlan configuration option in the server.conf), but I prefer to be sure

We support tagged mode, as you can see above, but the platform network (walt-net) must have its own dedicated LAN or VLAN.

> * Is it possible to de-couple the DHCP functionality of the server to let some third-party manage it? We may have access to a shared VLAN where there is already some dhcp server we have to keep in charge of the dhcp part (or at least, do not disturb the machines already in there);

No, this is currently not possible. DHCP is tightly coupled to WalT nodes booting procedure. As a result the DHCP service configuration file is maintained by walt server. Here is a short summary of nodes bootup procedure and how the DHCP service is involved.

1. On bootup, the network bootloader of the walt node sends a DHCP request to the server with a VCI field (vendor-class-identifier) set to `walt.node.<model>`, e.g. `walt.node.rpi-2-b` or `walt.node.pc-x86-64`.
2. The DHCP service returns an IP and runs a command `walt-dhcp-event` to notify main walt server process about this event (thanks to a specific directive in DHCP service configuration file).
3. If it is a new node, the main walt server process will verify that a default image for this `<model>` of node is ready, and otherwise download it from the docker hub. It will also have to mount this image if it was not used yet, expose it through NFS and maintain symlinks to make the node boot this image. And it will re-generate DHCP service configuration file in order to attach the given IP to this new node forever.

Other devices are also recorded in this configuration file when they are discovered. If you connect a switch on the walt network, it will get its own section in this file. If you connect other devices, and walt cannot recognize them (unknown mac address prefix and unknown DHCP VCI field), they will be recorded too, with type `unknown`.

> * Or, last, do you think it could be possible to use walt over a classic network without any dhcp at all? Could the vpn thing in preparation be used for that, maybe?

The "vpn node" I am preparing will establish its initial network connection through DHCP (at least in this first version), then establish the VPN, then start a kvm-based virtual machine connected to this VPN and that will be seen as the "walt node".
So yes, it should be able to boot on any regular network that provides a DHCP service and connectivity to walt VPN entrypoint.
However, I would like to highlight that it is probably better to stick to a regular walt platform network when possible. VPN nodes will have to traverse various networks, which may cause some problems (unexpected jitter for instance) in some experimental scenarios. And the fact the walt node is actually a virtual machine may prevent usability in some cases (we plan to attach USB ports of the host, but you will not be able to use other host peripherals).
Note that WalT was initially designed to run its platform network on a LAN for maximum isolation.

rgrunbla, 2019-11-13:
Thanks for the awesome answer !

I think we will go for the dedicated LAN, even if it means having a little bit less space for the experiments (it seems creating a new VLAN is a bit difficult in our university). The upside is we will get a manageable PoE switch and we will be able to control the nodes :)

Rémy

eduble, 2019-11-13:
> Thanks for the awesome answer !

You are welcome. :)

> I think we will go for the dedicated LAN, even if it means having a little bit less space for the experiments (it seems creating a new VLAN is a bit difficult in our university).

I must say we had to argue with network guys for several months here too... But we are happy with what we have now.

> The upside is we will get a manageable PoE switch and we will be able to control the nodes :)

OK. I suggest you select one of the switches we currently support (Netgear GS110TP is our favourite). Because some other models have non-standard MIBs and in this case WalT has to be modified to support them.

Etienne

closed 2019-11-14 by eduble