drakkar-lig/walt-python-packages #47: Walt OpenWRT Image

rgrunbla, 2019-12-13:
Hello,
I spent some time trying to get an openwrt image up and running for Walt. I finally managed to do so, but not without a few modifications to the walt scripts:

- As openwrt busybox is not statically compiled (unlike the one from debian), I had to modify walt-init to copy the dependencies in the chroot, in the `prepare_rootfs()` function. Could it be a good idea to integrate this to walt?

- I had to replace netcat with socat, because of a [bug in netcat](https://bugs.busybox.net/show_bug.cgi?id=12401) compiled with musl. I feel using socat for would be a better idea: socat has one implementation, while netcat has [many, with many inconsistencies](http://webcache.googleusercontent.com/search?q=cache:tEO1aUxbrBEJ:https://blog.i0i0.me/post/netcat.md&hl=fr&gl=fr&strip=1&vwsrc=0). Basically, that's just using `socat TCP:192.168.183.1:12345 system:$NC_SCRIPT` in walt-rpc, using ` socat STDIO TCP-LISTEN:123456,reuseaddr` in walt-net-service, as well as adding socat to the IMAGE_SCRIPTS in walt-init.

- I had a small problem with busybox date in the walt-clock-sync script: the timestamp since epoch was not accepted. I fixed it by using `busybox date -s "$(busybox date -d @$ts "+%Y-%m-%d %H:%M:%S")"` in place of the original command. I don't know if this is a general problem or just a problem with the openwrt busybox.

That's what I had to change on the walt-side to make it work.
All of the build process is detailed there: https://github.com/rgrunbla/openwrt-walt-image (only for apu4c4 right now).

Thanks again for this project,
Rémy

eduble, 2019-12-13:
Hi Rémy,
Thanks for your effort.

> As openwrt busybox is not statically compiled (unlike the one from debian), I had to modify walt-init to copy the dependencies in the chroot, in the prepare_rootfs() function. Could it be a good idea to integrate this to walt?

Sure.

> I had to replace netcat with socat, because of a bug in netcat compiled with musl. I feel using socat for would be a better idea: socat has one implementation, while netcat has many, with many [...]

We have chosen `nc` because it is available in busybox, thus we did not need more external software on the image. And I think we cannot break compatbility with all images that just have `nc`, and no `socat`. But we could handle both. Maybe, use `socat` if we have it, `nc` otherwise. Is `socat` statically compiled? Or I guess you had to reuse the technique for copying binary dependencies? 

> I had a small problem with busybox date in the walt-clock-sync script: the timestamp since epoch was not accepted. I fixed it by using busybox date -s "$(busybox date -d @$ts "+%Y-%m-%d %H:%M:%S")" [...]

We did not see this problem ourselves. But if your command is more robust, we can replace it.

The code repository you mentioned is only for the walt image, right? Did you record somewhere else those modifications on walt server side?

Thanks,
Etienne

rgrunbla, 2019-12-13:
> Maybe, use socat if we have it, nc otherwise. Is socat statically compiled? Or I guess you had to reuse the technique for copying binary dependencies?

Well, I think it worked out of the box because it shares the same dependencies as busybox.

As for the "dependency detection", I did it by hand, but I can probably work out a simple script to get them from ldd and copy them in the chroot.

For the modifications, right now I didn't put them anywhere, but here are the patches. For the walt-init, I harcoded the libraries from ldd by hand, so of course it's not usable as such.

## walt-init

```
# diff -u walt-init /usr/local/lib/python2.7/dist-packages/walt/server/threads/main/images/walt-init
--- walt-init	2019-12-13 09:42:42.927817810 +0100
+++ /usr/local/lib/python2.7/dist-packages/walt/server/threads/main/images/walt-init	2019-10-28 11:00:09.240172256 +0100
@@ -6,16 +6,12 @@
 ROOTFS='/tmp/rootfs'
 ROOTFS_BUSYBOX_APPLETS="
     sh ls timeout reboot sleep nc chroot
-    mktemp mkfifo sed cat chmod rm date
+    mktemp mkfifo sed cat
 "
 IMAGE_SCRIPTS="
     /bin/walt-nfs-watchdog /bin/walt-net-service
     /bin/walt-notify-bootup /bin/walt-rpc
-    /bin/walt-clock-sync /usr/bin/socat
-"
-LIBS="
-   /lib/ld-musl-i386.so.1 
-   /lib/libgcc_s.so.1 
+    /bin/walt-clock-sync
 "
 
 admin_sh()
@@ -34,7 +30,7 @@
 {
     echo '>>> Issue detected!               <<<'
     echo 'Will reboot... (or press <ENTER> now for a shell)'
-    busybox timeout 5 busybox head -n 1 && admin_sh || busybox reboot -f
+    busybox timeout -t 5 busybox head -n 1 && admin_sh || busybox reboot -f
 }
 
 trap reboot_or_sh EXIT
@@ -65,13 +61,8 @@
 {
     mkdir $ROOTFS
     cd $ROOTFS
-    mkdir -p tmp bin dev mnt/nfsroot lib
+    mkdir -p tmp bin dev mnt/nfsroot
     cp "$(which busybox)" bin/
-    ldd bin/busybox
-    for lib in $LIBS
-    do
-        cp $lib lib/
-    done
     for image_script in $IMAGE_SCRIPTS
     do
         cp $image_script bin/
```
## walt-clock-sync

```
# diff -u walt-clock-sync /usr/local/lib/python2.7/dist-packages/walt/server/threads/main/images/walt-clock-sync 
--- walt-clock-sync	2019-12-12 14:57:12.259356205 +0100
+++ /usr/local/lib/python2.7/dist-packages/walt/server/threads/main/images/walt-clock-sync	2019-05-16 16:47:13.000000000 +0200
@@ -2,7 +2,8 @@
 
 # get timestamp from server (format epoch)
 ts=$(/bin/walt-rpc sync_clock)
+
 # set local clock
-busybox date -s "$(busybox date -d @$ts "+%Y-%m-%d %H:%M:%S")"
+busybox date -s "$(busybox date -D "%s" -d $ts "+%Y-%m-%d %H:%M:%S")"
 
 ```

## walt-rpc

 ```
# diff -u walt-rpc /usr/local/lib/python2.7/dist-packages/walt/server/threads/main/images/walt-rpc 
--- walt-rpc	2019-12-12 14:41:11.989707651 +0100
+++ /usr/local/lib/python2.7/dist-packages/walt/server/threads/main/images/walt-rpc	2019-05-16 16:47:13.000000000 +0200
@@ -15,7 +15,7 @@
 EOF
 chmod +x $NC_SCRIPT
 
-socat TCP:192.168.183.1:12345 system:$NC_SCRIPT 2>&1 | {
+nc %(server_ip)s %(walt_server_rpc_port)s -e $NC_SCRIPT 2>&1 | {
     read response_word1 response_rest
     word2=$(echo -n "$response_rest" | sed -e 's/.$//')
     if [ "$response_word1" = "('RESULT'," ]
```

 
## walt-net-service

```
# diff -u walt-net-service /usr/local/lib/python2.7/dist-packages/walt/server/threads/main/images/walt-net-service 
--- walt-net-service	2019-12-12 15:21:01.280217146 +0100
+++ /usr/local/lib/python2.7/dist-packages/walt/server/threads/main/images/walt-net-service	2019-05-16 16:47:13.000000000 +0200
@@ -23,7 +23,7 @@
 # to communicate with the server
 while [ 1 ]
 do
-    socat STDIO TCP-LISTEN:123456,reuseaddr
+    nc -l -p %(walt_node_net_service_port)s
 done < $fifo_out > $fifo_in &
 
 # pipe fd 5 to standard output of nc (we will read
```

Rémy

eduble, 2019-12-16:
OK, thanks for the input.
I am planning to work on this at the beginning of january, together with other server code changes related to the issue #43 you reported. I am not sure yet if I will release a new version at this time, or if I will finalize what I am working on right now, in order to have it included... (something that could better mitigate network latency for VPN nodes). I will keep you updated at this time.

rgrunbla, 2019-12-17:
Ok! There is nothing urgent on our side, so feel really free to deal with theses tickets (close them, …) in the way you prefer.

eduble, 2020-09-08:
Closing this, since the long awaited [version 5](https://listes.univ-grenoble-alpes.fr/sympa/arc/walt-users/2020-09/msg00000.html) that we just released fixes those issues.

closed 2020-09-08 by eduble