drakkar-lig/walt-python-packages #49: Problem with "walt device tree" output

Warxo, 2020-02-24:
Hello,
I'm trying to connect my nodes to Poe switch Netgear GS110TP. I followed your setup but I'm having some problems.

This may be linked to issue #45 

I can see that the node connected to the switch works fine but the following commands doesn't seem to detect any node.
```
walt device rescan
walt device tree 
```

Here you can see the results:
```
root@localhost:~# walt device rescan
Querying walt server               OK
Querying switch                    OK
root@localhost:~# walt device tree

-- switch --
 └─[...] ~> no nodes there

note: this view comes from last network scan, issued 7 seconds ago (use 'walt device rescan' to update).

tips:
- use 'walt device tree --all' to see all devices detected
- use 'walt device show' for device details
- use 'walt device admin <switch>' to let WalT explore forbidden switches
- use 'walt device rescan' to update
root@localhost:~# walt device rescan
Querying walt server               OK
Querying switch                    OK
root@localhost:~# walt device tree

-- switch --
 └─[...] ~> no nodes there

note: this view comes from last network scan, issued 7 seconds ago (use 'walt device rescan' to update).

tips:
- use 'walt device tree --all' to see all devices detected
- use 'walt device show' for device details
- use 'walt device admin <switch>' to let WalT explore forbidden switches
- use 'walt device rescan' to update
```
Here is the result of my device show command. At this moment only the node called "ap" is linked to the Poe switch.
```
root@localhost:~# walt device show

The WalT network contains the following devices:

name         ip             mac                type    
-----------  -------------  -----------------  ------  
ap           192.168.183.3  00:0d:b9:51:a7:b0  node    
client       192.168.183.4  00:0d:b9:51:a7:b4  node    
monitor      192.168.183.2  00:0d:b9:51:a5:5c  node    
walt-server  192.168.183.1  52:54:00:36:6d:b5  server  
switch       192.168.183.5  38:94:ed:d0:22:61  switch  

tips:
- use 'walt device tree' for a tree view of the network
- use 'walt device forget <device_name>' to make WalT forget about an obsolete device

WalT also detected the following devices but could not detect their type:

name             ip              mac                type     
---------------  --------------  -----------------  -------  
-7------8---140  10.100.205.140  94:f1:28:8f:5d:c0  unknown  
localhost        140.77.67.27    00:1d:09:0c:bb:c5  unknown  
mon-pc           192.168.183.6   dc:4a:3e:60:33:60  unknown  

Use 'walt device admin <name>' to fix this.
```
As you can see I've got a weird line "-7------8---140  10.100.205.140  94:f1:28:8f:5d:c0  unknown " and I don't know when and why it appears. I can't forget it.

Thanks in advance,
Paul

eduble, 2020-02-26:
Hi Paul,

Is the device named "switch" your new PoE switch? Or is it another one?
Here are a few tips:
- LLDP detection requires that both the node and the switch have LLDP support. The netgear switch has it enabled by default. But does your node run a LLDP daemon? If not, a simple way to have it detected is to temporarily boot the default image on it. Note that the switch may need a few minutes to have the node detected through LLDP. Once it is detected, you may reboot another image on it, the position of the node in the topology will be saved (but if you change topology and your image has no LLDP daemon, then of course the change will not be detected).
- walt does not communicate with switches by default (since this usually requires agreement from network admins). You have to type command `walt device admin <switch-device-name>` to allow walt to perform SNMP requests on it. You will have to answer questions about what walt is allowed to do (LLDP and/or PoE requests).
- walt may not be able to detect that a device is a switch. Again, command `walt device admin <switch-device-name>` can fix this case too.
- according to `https://hwaddress.com/oui-iab/94-F1-28/` the device with a strange name is a Hewlett Packard device. Do you remember an HP device was ever connected somewhere on your walt network?
- When interrogating a LLDP-capable switch (or the LLDP daemon running on server), LLDP information about each neighbor includes mac address, ip address, switch port and sometimes a "device name". If this "device name" is provided, and the detected device is new, walt records it as the device name in walt database. However, it has to ensure that this device name does not contain special chars. Thus it replaces all unexpected chars (not a letter, not a number, not a dash) with a dash. (The relevant code is at https://github.com/drakkar-lig/walt-python-packages/blob/master/server/walt/server/threads/main/devices/topology.py#L311) This probably explains the strange name you get. But the fact you cannot forget it is an issue (is it because it starts with a dash?).
- If you type `walt advanced sql` and `select * from topology` you should get information about the LLDP links that were detected. (Association between device mac address and name can be found in table `devices`.) This is the information walt uses to display the tree.

Let me know if some of these tips can help you.
Etienne

Warxo, 2020-02-28:
Hello,

> Hi Paul,
> 
> Is the device named "switch" your new PoE switch? Or is it another one?

Yes

> Here are a few tips:
>
>     * LLDP detection requires that both the node and the switch have LLDP support. The netgear switch has it enabled by default. But does your node run a LLDP daemon? If not, a [...]

It's maybe my laptop.

> * When interrogating a LLDP-capable switch (or the LLDP daemon running on server), LLDP information about each neighbor includes mac address, ip address, switch port and sometimes a "device name". [...]

Thank you for the tips, I will try to install the lldp deamon on my node and check if it see new topology.
I will keep you in touch when I be back.

Paul

eduble, 2020-03-02:
About the device with a strange name, you can add token `--` to your walt command to indicate the end of options:
```
$ walt device rename -- '-7------8---140' hp-device
```
This will prevent the device name to be interpreted as an option, thus the command should work.

eduble, 2020-09-08:
Fixed by 4e2199a526f5350752418a9726d3c31eed75c92a.

closed 2020-09-08 by eduble