drakkar-lig/walt-python-packages #50: Bug with walt-server when dealing with duplicated layers ?

Warxo, 2020-07-20:
Hello,

I managed to break my walt-server installation with one of my image.

The logs (full: [journalctl_p1.txt](https://github.com/drakkar-lig/walt-python-packages/files/4948520/journalctl_p1.txt)) contain this kind of messages :

```
Jul 20 14:20:41 localhost walt-server-daemon[2056]: Exception in thread server-main:
[...]
Jul 20 14:20:41 localhost walt-server-daemon[2056]:   File "/usr/local/lib/python2.7/dist-packages/walt/server/threads/main/mydocker.py", line 134, in image_mount
Jul 20 14:20:41 localhost walt-server-daemon[2056]:     mount('-t', 'aufs', '-o', branches_opt, 'none', mount_path)
[...]
Jul 20 14:20:41 localhost walt-server-daemon[2056]: ProcessExecutionError: Command line: [u'/bin/mount', u'-t', u'aufs', u'-o', u'br=/var/lib/walt/images/paulgrangette/pc-x86-64-client2_latest/diff=rw:/var/lib/docker/aufs/diff/5fccfa02e867d5976cc555aac4377f2208f9403556e91fb5cd6d012c5b3d1732=ro+wh:/var/lib/docker/aufs/diff/5fccfa02e867d5976cc555aac...
```

Looking at dmesg, aufs complains about some layers being duplicated:

```
[ 1030.111718] aufs test_add:216:mount[2204]: /var/lib/docker/aufs/diff/5fccfa02e867d5976cc555aac4377f2208f9403556e91fb5cd6d012c5b3d1732 duplicated
```

The image that triggers this message is available at https://hub.docker.com/repository/docker/paulgrangette/pc-x86-64-client2

It seems that it contains duplicated layers, triggering some bug in walt-server.

I managed to circumvent this problem by editing the mydocker.py file and removing the duplicates in the list `branches[:AUFS_BR_MOUNT_LIMIT]`: https://github.com/drakkar-lig/walt-python-packages/blob/e5068b8d0b3addb1fa9177359337fe35b0ae47cb/server/walt/server/threads/main/mydocker.py#L139

            # we can mount up to AUFS_BR_MOUNT_LIMIT branches at once
            orig=branches[:AUFS_BR_MOUNT_LIMIT]
            new=[]
            already_seen=set()
            for item in orig:
                if item not in already_seen:
                    new.append(item)
                    already_seen.add(item)
                else:
                    print("skipping %s"%item)
            #branches_opt = 'br=' + ':'.join(branches[:AUFS_BR_MOUNT_LIMIT])
            branches_opt = 'br=' + ':'.join(new)

Is this a problem with my image (or image generation) or is it a problem with walt-server ?

Thanks,
Paul

eduble, 2020-07-21:
Hi Paul,
Being able to export over NFS the content of a docker image is a very important feature in WalT. However, when I wrote this code, there was no support in docker tooling to make just this mount operation, so I had to analyse the internal storage structure docker uses to store image layers, and write this code.
The problem is that this internal storage structure has evolved. To be more precise, docker has provided other kinds of storage backends. At this time it was based on aufs union filesystem. It is now deprecated because aufs has never been merged into linux kernel, and another union filesystem called 'overlay' or 'overlayfs' was merged. So the default storage backend for newer docker installations is based on overlayfs, and aufs installation on a newer OS is more and more complicated (especially when we want the NFS export ability).
Some time ago, I found a tool called `buildah`. It allows to build container images (docker format included), and it provides a subcommand `buildah mount` that performs what we want.
So I started to work on a new branch called `buildah-podman` in this repository, where image management is now based on `buildah` and its companion tool `podman`. In this version, the image repository of the walt server (managed by `buildah` and `podman`) is different from the image repository of `docker`. But you can still use docker to build your images, and then import them into walt own repository by using `walt image search` and `walt image clone` (such an image will be displayed with a clone URL starting with "docker:").
I would like to publish a new version of walt based on this branch shortly. It was planned before the end of this month, but I still have a few things to check, and I will have holydays, so it might actually be at the end of august.
Since the code you got a problem with is obsolete, I was wondering if you would like to beta test this new version and report to me any problem you might notice? If yes, I will explain you how to update (this is a major update, with python 2 to 3 transition, etc.).

Warxo, 2020-07-21:
Hello,

Thank you for the answer.
Unfortunately I will not be able to test the beta of this new version because my internship finish very soon and I will not have the time. 

I will let you close the issue when you want. 
I will keep you in touch if I meet any other bug.
Paul

eduble, 2020-07-21:
OK I understand. If anyone in your team is interested by this preview, let me know. If you are using some features we do not use much here, this might allow to improve a few more things before release.
Did the patch you made allow to mount the image correctly?

Warxo, 2020-07-21:
Yes I don't have an error since I made this changes.

eduble, 2020-07-21:
OK thanks. I will let the issue open until next release.

eduble, 2020-09-08:
Closing this, since new [version 5](https://listes.univ-grenoble-alpes.fr/sympa/arc/walt-users/2020-09/msg00000.html) was just released and relies on a different image storage.

closed 2020-09-08 by eduble