drakkar-lig/walt-python-packages #58: Wrong interface used for snmp ?

rgrunbla, 2021-02-27:
Hi,

I use a netgear manageable switch, and added it as a switch in walt. Still, it looks like ```walt device rescan``` is not working as expected:

```
# walt device rescan
Querying walt server               OK
Querying switch-44fd40             FAILED (while trying to probe SNMP variant)
```

Looking at the output of "show neighbors" using ```lldpcli``` on the walt server, everything looks fine.

The main problem is that the packets seem to be sent on the wrong interface: eth0 (while the walt-net interface is eth1):

```
root@localhost:~# cat /etc/walt/server.conf
{
    # network configuration
    # ---------------------
    "network": {
        # platform network
        "walt-net": {
            "raw-device": "eth1",
            #"vlan": <nnnn>,
            "ip": "192.168.192.1/24"
        }
    }
}
```

Output of some tcpdump running in a tmux:

```

root@localhost:~# tcpdump 'port 161 or 162' -i eth1
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth1, link-type EN10MB (Ethernet), capture size 262144 bytes

[empty]

─────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────
root@localhost:~# tcpdump 'port 161 or 162' -i eth0
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on eth0, link-type EN10MB (Ethernet), capture size 262144 bytes
17:23:15.460932 IP dhcp-67-209.ixxi.ens-lyon.fr.41542 > M7-SW-SR8-A-138.ens-lyon.fr.snmp:  C="private" GetBulk(30)  N=0 M=40 .iso.0.8802.1.1.2.1.4.1.1.4
17:23:18.466099 IP dhcp-67-209.ixxi.ens-lyon.fr.41542 > M7-SW-SR8-A-138.ens-lyon.fr.snmp:  C="private" GetBulk(30)  N=0 M=40 .iso.0.8802.1.1.2.1.4.1.1.4
17:23:21.471190 IP dhcp-67-209.ixxi.ens-lyon.fr.41542 > M7-SW-SR8-A-138.ens-lyon.fr.snmp:  C="private" GetBulk(30)  N=0 M=40 .iso.0.8802.1.1.2.1.4.1.1.4
17:23:24.478795 IP dhcp-67-209.ixxi.ens-lyon.fr.41542 > M7-SW-SR8-A-138.ens-lyon.fr.snmp:  C="private" GetBulk(33)  N=0 M=40 E:11863.6.35.1.2.2.1.3
17:23:27.483865 IP dhcp-67-209.ixxi.ens-lyon.fr.41542 > M7-SW-SR8-A-138.ens-lyon.fr.snmp:  C="private" GetBulk(33)  N=0 M=40 E:11863.6.35.1.2.2.1.3
17:23:30.488928 IP dhcp-67-209.ixxi.ens-lyon.fr.41542 > M7-SW-SR8-A-138.ens-lyon.fr.snmp:  C="private" GetBulk(33)  N=0 M=40 E:11863.6.35.1.2.2.1.3

```

What could cause the wrong interface to be used ?

Thanks,
Rémy

eduble, 2021-03-01:
Hi Rémy,
Walt just uses the IP of the switch to reach it.
Could you run the following:
```
$ walt device show | grep switch-44fd40
```
Your switch might have an IP in another network (outside walt-net), thus SNMP queries are sent to your IP gateway (thus through eth0).

If your switch is physically in `walt-net` network, then it should be configured to get its IP address with DHCP. Thus it will get an address from walt server, and this IP will belong to `walt-net`. Just take care to reboot the switch after walt server is up and running, otherwise the switch may have no answer to its DHCP requests and fall back to using a default IP.

Unfortunately we have less control on the behavior of switches, comparing to nodes. The bootloader on nodes is configured to make them reboot until they get a valid IP, but we cannot do the same with switches...

For more info:
The following code creates the SNMP proxy object used fro getting switch neighbors: https://github.com/drakkar-lig/walt-python-packages/blob/master/server/walt/server/threads/main/devices/topology.py#L386
This proxy object internally relies on the `snimpy` library.

rgrunbla, 2021-03-01:
Indeed, you're completely right, thanks !

```
root@localhost:~# walt device show | grep switch-44fd40
switch-44fd40  10.100.205.138  98:f2:b3:44:fd:40  switch
root@localhost:~# ip r get 10.100.205.138
10.100.205.138 via 140.77.67.1 dev eth0 src 140.77.67.209 uid 0
    cache
root@localhost:~#
```

I'll try to find a way to reboot the switch remotely, or add a route by hand to overwrite the use of the default route.

Thanks again,
Rémy

closed 2021-03-01 by rgrunbla