drakkar-lig/walt-python-packages #64: Infinite loop when postgresql is not running

audeoudh, 2021-10-04:
https://github.com/drakkar-lig/walt-python-packages/blob/b778992e241d54b684f54715d83c4aff98a01db7/server/walt/server/postgres.py#L14-L19

Those lines runs infinitely when PostgreSQL is not running.  `self.conn` never gets assigned, `psycopg2.connect()` always raise `OperationalError`.

OK, that's a problem, PostgreSQL is expected to run, so we may exit with an exception—but we shouldn't enter an infinite loop.

Proposed modifications (for discussion), replace the whole loop by:

```python
if self.conn is None:
    try:
        # Do catch exception here to create DB and users if there does not exist
        self.conn = psycopg2.connect(...)
    except psycopg2.OperationalError:
        self.create_db_and_user()
        # Do not catch exception here to let the user know if something happens bad
        self.conn = psycopg2.connect(...)
```

eduble, 2021-10-05:
OK.
So this means `self.create_db_and_user()` does not raise an exception in this case?

audeoudh, 2021-10-05:
Not in my test setup.

commit 2e59f8d7 (2021-10-05): Fix infinite loop when postgresql is not running

Fix #64
commit 4e47e4bc (2021-10-21): Fix infinite loop when postgresql is not running

Fix #64
closed 2023-03-16 by eduble