drakkar-lig/walt-python-packages #70: Enhance walt-vpn-endpoint: forward SSH file descriptors to the WalT VPN server

audeoudh, 2021-10-18:
Using `SCM_RIGHTS`, we should be able to pass the two file descriptors of the SSH daemon from the WalT VPN endpoint to the WalT VPN server through a UNIX socket file.  Thus, WalT VPN endpoint will be used only at startup, then terminates, avoiding resources consumption and back and forth data copies with the kernel.

audeoudh, 2021-10-18:
Not even that.

`walt-vpn-endpoint` would forward its "stdin" and "stdout" file descriptors through the UNIX socket, then terminates.  `walt-vpn-server` receives the file descriptors through the UNIX socket and is able to handle errors itself.

Thus, there is no running `walt-vpn-endpoint`.  The only need we have of it is to connect the `walt-vpn-server` (which is running as root) to the SSH endpoint itself (which is running as the walt-vpn user).

See `nc -F` and `ssh -o ProxyUseFdPass=yes` for similar techniques.  Also [that article](https://www.gabriel.urdhr.fr/2016/08/07/openssh-proxyusefdpass/) which is testing it.

eduble, 2021-11-23:
I finally implemented it the other way, for a better scalability and simplified code:
`walt-vpn-server` opens the tap then sends it to `walt-vpn-endpoint`, then `walt-vpn-endpoint` handles the transfers between the ssh channel and the tap.
This is more scalable because each client gets its own `walt-vpn-endpoint` process, compared to handling the heavy processing of all clients on `walt-vpn-server` side. 
The C code is also simpler and could be refactored because after the initialization `walt-vpn-endpoint` and `walt-vpn-client` do basically the same thing (transfers between a tap and an ssh channel).
This issue will be marked solved when the relevant commit will be pushed to master.

commit 6564fbdd (2021-11-23): walt-vpn: improve scalability & security

walt-vpn-server now just creates a tap and sends its file descriptor
to walt-vpn-endpoint through socket message ancilliary data.
Next walt-vpn-endpoint manages the transfers between the tap and the
ssh channel.
This is more scalable than letting the server manage all clients alone.
Moreover, walt-vpn-endpoint and walt-vpn-client now share most of the
same C code for these transfers between the tap and the ssh channel.

On the security front, the VPN Unix socket is now only readable /
writable by the walt-vpn user. Previously, this user was created
whe [... cut]
commit 512b001e (2023-03-16): walt-vpn: improve scalability & security

walt-vpn-server now just creates a tap and sends its file descriptor
to walt-vpn-endpoint through socket message ancilliary data.
Next walt-vpn-endpoint manages the transfers between the tap and the
ssh channel.
This is more scalable than letting the server manage all clients alone.
Moreover, walt-vpn-endpoint and walt-vpn-client now share most of the
same C code for these transfers between the tap and the ssh channel.

On the security front, the VPN Unix socket is now only readable /
writable by the walt-vpn user. Previously, this user was created
whe [... cut]
closed 2023-03-16 by eduble