drakkar-lig/walt-python-packages #72: Possible `UnboundLocalError` for variable `username`

audeoudh, 2022-01-03:
https://github.com/drakkar-lig/walt-python-packages/blob/007b860a99d46c178ae0ddff62672af8ca37d930/server/walt/server/threads/main/images/store.py#L179-L182

These lines may raise a `UnboundLocalError` for variable `username`.

Bug added in 938eac46.

Example of command to run in this bug : `walt image shell docker:waltplatform/rpi-alpine` with `rpi-alpine` not in the user's image set.

eduble, 2022-01-04:
> Example of command to run in this bug : `walt image shell docker:waltplatform/rpi-alpine` with `rpi-alpine` not in the user's image set.

Did you mean `walt image clone docker:waltplatform/rpi-alpine`?

eduble, 2022-01-04:
I think we just have to replace line 170 here:

https://github.com/drakkar-lig/walt-python-packages/blob/007b860a99d46c178ae0ddff62672af8ca37d930/server/walt/server/threads/main/images/store.py#L170

Replace it with:
```
fullname, username, _ = parse_image_fullname(image_name)
```

And import `parse_image_fullname` from `walt.server.threads.main.images.image`.

audeoudh, 2022-01-04:
> Did you mean `walt image clone docker:waltplatform/rpi-alpine`?

No, no, `walt image shell`.  I *hoped* that it would clone it previously, but did not check if it is implemented thus. :-)  Even if it isn't, it should not fail with `UnboundLocalError`.

> I think we just have to replace line 170 […] with:
> 
> ```
> fullname, username, _ = parse_image_fullname(image_name)
> ```

Seems OK to me.

Just to check, is the `docker:` prefix already stripped when entering this function?

eduble, 2022-01-04:
> Just to check, is the docker: prefix already stripped when entering this function?

Not 100% sure, but I think yes.

> No, no, `walt image shell`. I _hoped_ that it would clone it previously, but did not check if it is implemented thus. :-)

I see.

> Even if it isn't, it should not fail with `UnboundLocalError`.

Obviously.

commit 9a9d8d17 (2022-01-13): get_user_image_from_name() fix: disallow image fullname as input

For the "walt node cp ... booted-image" feature, function
get_user_image_from_name() had been modified to allow specifying
an image fullname instead of a user image name as input.
This was not a good idea and caused issues (cf. issue #72).

This commit reverts this change and handles the "booted-image" case
before calling this function.

Fix #72.
commit 40a33c2e (2023-03-16): get_user_image_from_name() fix: disallow image fullname as input

For the "walt node cp ... booted-image" feature, function
get_user_image_from_name() had been modified to allow specifying
an image fullname instead of a user image name as input.
This was not a good idea and caused issues (cf. issue #72).

This commit reverts this change and handles the "booted-image" case
before calling this function.

Fix #72.
closed 2023-03-16 by eduble