drakkar-lig/walt-python-packages #83: Improve layer-2 network topology discovery

eduble, 2022-11-17:
We currently use LLDP, but this requires that both the switch and the device support it.
In order to detect where a device is (probably) connected, we could also read the [dot1qTpFdbPort OID from Q-BRIDGE-MIB](https://cric.grenoble.cnrs.fr/Administrateurs/Outils/MIBS/?oid=1.3.6.1.2.1.17.7.1.2.2.1.2):

```
root@milo:~# python3
>>> from walt.server.processes.blocking import snmp
>>> from walt.server.processes.blocking.snmp.mibs import load_mib, unload_mib
>>>
>>> p = snmp.Proxy('192.168.172.133', {'community': 'XXXXX', 'version': 2}, lldp=True)
>>> p.lldp.get_neighbors()
{4: {'mac': '6c:b0:ce:18:d5:6c', 'ip': '192.168.172.10', 'sysname': ''}, 3: {'mac': 'b8:27:eb:60:51:d4', 'ip': '192.168.172.143', 'sysname': 'rpi3bp-468-J'}}
>>>

>>> load_mib('Q-BRIDGE-MIB')
>>> dict(p.snmp.dot1qTpFdbPort)
{(<Unsigned32: 1>, <String: 0:d:b9:45:f8:90>): <Integer: 4>, (<Unsigned32: 1>, <String: 0:50:56:28:95:9f>): <Integer: 4>, (<Unsigned32: 1>, <String: 18:fd:74:6b:e:64>): <Integer: 4>, (<Unsigned32: 1>, <String: 18:fd:74:6b:e:b8>): <Integer: 4>, (<Unsigned32: 1>, <String: 18:fd:74:6b:f:8a>): <Integer: 4>, (<Unsigned32: 1>, <String: 3a:39:5a:0:e0:db>): <Integer: 4>, (<Unsigned32: 1>, <String: 44:94:fc:8e:fa:f>): <Integer: 13>, (<Unsigned32: 1>, <String: 58:bf:25:e7:94:b7>): <Integer: 4>, (<Unsigned32: 1>, <String: 58:bf:25:e7:b0:9f>): <Integer: 2>, (<Unsigned32: 1>, <String: 58:bf:25:e7:b3:a7>): <Integer: 1>, (<Unsigned32: 1>, <String: 6c:b0:ce:18:d5:6e>): <Integer: 4>, (<Unsigned32: 1>, <String: 70:76:ff:3:6:aa>): <Integer: 4>, (<Unsigned32: 1>, <String: b8:27:eb:60:51:d4>): <Integer: 3>, (<Unsigned32: 1>, <String: f8:73:a2:2e:c8:0>): <Integer: 4>}
>>> 
```

We get a dict() with the form `<key> = (<vlan-id>, <mac>)` and `<value> = <port-index>`.
This is the switch forwarding table, so mac addresses may not be immediate neighbors.
However, if we have multiple mac addresses on a given port (such as port 4 here), we know this port is connected to another switch. Thus a probably good estimate can be obtained by doing the following steps:  
1) we do this query on all known switches
2) we ignore results about ports associated to multiple mac addresses
3) we ignore mac addresses for which we already have an LLDP result 
4) we look for mac addresses referenced only once in the network after the previous filterings

Note : dot1qTpFdbPort only works for VLAN-capable switches. For non-VLAN-capable switches we should use [dot1dTpFdbPort OID from BRIDGE-MIB](https://cric.grenoble.cnrs.fr/Administrateurs/Outils/MIBS/?oid=1.3.6.1.2.1.17.4.3.1.2) instead.
I do not have non-VLAN-capable switches to try. The returned directory is probably with the form `<key> =<mac>` and `<value> = <port-index>`, but this remains to be confirmed. In the meantime, we should restrict this lookup to dot1qTpFdbPort.

eduble, 2022-11-17:
As a reminder to myself: I should also test this on our Avaya switches.

audeoudh, 2022-11-21:
Does the port numbers reported by the forwarding table matches the ports numbers reported by LLDP or by PoE? If I remember correctly, these two differ.

eduble, 2022-11-21:
Hi Henry, you are true and I don't know, I did not get this far yet. This will have to be investigated.

commit 095dcfb9 (2022-11-24): network discovery: use Q-BRIDGE MIB to locate non-LLDP devices

Fixes #83.
commit f4ff4d49 (2022-11-25): network discovery: use Q-BRIDGE MIB to locate non-LLDP devices

Fixes #83.
closed 2023-03-16 by eduble